STMicroelectronics/BlueSTSDK_Android #1: Example app - BLE connection issue

ivaninastoilova, 2016-08-18:
Ciao! 

I am developing a simple app using your sensor and I have been using your sample app as a guide on how to connect with the sensor. However, I am having significant issues establishing a sustainable connection with the sensor for over 2 minutes. The bigger problem is that once the sensor disconnects, I cannot get it to reconnect again. I have noticed that the Example app cannot reconnect either and I was wondering whether it is possible to gain access to the code of the ST WeSU official app in any way? 

Grazie mille.

Best Regards,
Iva

GiovanniVisentini, 2016-08-22:
Ciao Iva,

The example app has the ability to re connect, using the NodeStateListener [NodeContainerFragment.java#L117](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Android/blob/master/BlueSTExample/src/main/java/com/st/BlueSTSDK/Example/NodeContainerFragment.java#L117) . However, it doesn't re enable the notification.
In your app you should have NodeStateListener that enables the needed notification when the node enter in a connect state.

If you are using the Wesu sensor and Android 6, we have a known problem, since the device is setting a connection interval that Android 6 does not support anymore. 
You can set a bigger connection interval using the debug console with the command:

`!bleconintv 10` for set the connection interval to 12.5ms or the safest `!bleconintv 30` for set the connection interval to 30ms

Best Regards,
Giovanni

ivaninastoilova, 2016-08-26:
Ciao Giovanni,

Thank you for the quick response. I have tried the change in the connection interval this week and the application still cannot reconnect. I also want to clarify that the Example application does not reconnect as well - I have tested it by disconnecting the sensor (taking it outside the range of the BLE connection) and when returned within the range, the sensor does not automatically connect, rather it gets stuck in discovery. I can provide screenshots, but I am certain you can test it yourself.

I have also followed your other suggestion - enabling the notifications, but again, to no success.

Do you have any other recommendations?

Thank you.

Best Regards,
Iva

GiovanniVisentini, 2016-08-29:
Ciao Iva,

I tested the Wesu with the firmware 1.0.01, and the example app  with a nexus9, nexus 6p, Sony z3c with android 6 and an S4 with android 4.4.
With my devices I have a disconnection when I enable more than 4 notifications but the re connection is almost instantaneous, and after the re connection you have to re enable the notification..

Which device are you using?