STMicroelectronics/BlueSTSDK_Android #2: Battery feature

adelorenz, 2017-02-13:
Hi,
I'm unable to retrieve battery data from a SensorTile using this SDK.
The battery feature doesn't appear in the list of available features, but the BlueMS app is able to retrieve and to show such a feature.

I've tried to manually obtain the data with
`node.getFeature(FeatureBattery.class);`
but it returns a null object.

I'm able to retrieve all others data (mag, gyro, acc, ...).
I'm forgetting something?

GiovanniVisentini, 2017-02-14:
Hi Andrea,

I tested with Bluemicrosystem2  v2.2 and  BlueSTSDKExample application and I'm able to see the battery data.

- Which firmware are you using?
- Do you call getFeature after the connection has been completed?

adelorenz, 2017-02-15:
Hi Giovanni,
i'm using  Bluemicrosystem2 v2.2.

Here is a portion of my code:
```
private Node.NodeStateListener mNodeStateListener = new Node.NodeStateListener() {
        @Override
        public void onStateChange(final Node node, Node.State newState, Node.State prevState) {
            if ((newState == Node.State.Connected)) {
                List<Feature> features = node.getFeatures();
                for (final Feature feature : features) {
                    node.enableNotification(feature);
                    // NO BATTERY FEATURE HERE
                }
                Feature battery = node.getFeature(FeatureBattery.class); // NULL
            } else if ((newState == Node.State.Unreachable || newState == Node.State.Dead || newState == Node.State.Lost)) {
                // DO STUFF HERE
            } else if (newState == Node.State.Disconnecting) {
                // DO STUFF HERE
            }
        }
    };
```

dule-hwp, 2017-02-17:
Hi Giovanni, 

I am using SensorTile and example Android app and want to get the FeatureCompass readings. I guess my problem is the same as for Andrea, tried pretty much the same as he did with the same outcome.

I think the main problem i am facing is that i cant change the advertised data being received in **BluetoothAdapter.LeScanCallback.onLeScan** 
`public void onLeScan(final BluetoothDevice device, int rssi, byte[] advertisedData) {`
callback method. So, by default, after all Advertise parsing is done, feature mask built is 0x4F40400. And thats why we dont see the rest of the features in the node features list (we can only see as many 1s there are, hence 7 features). So, i assume if we can set other featureMask bits to 1s we can expose other features? 

I think that this is done when you use license manager (using sample app from Google Play, which is different from the one we have code in this repo for some reason..?) and add osxMotionFX for example. Then when i get back to the example app hosted on this repo, i can see that feature mask is changed to 0x4F40500 and i have one more feature enabled (Sensor Fusion Compact feature) as expected. This makes me think i can somehow change what is advertised from the SensorTile, but since we dont have access to the code from the play store I cant figure out how is done. Do you have idea how this can be done?

Thank you for your time, and sorry about long post..

GiovanniVisentini, 2017-02-17:
Hi Dusan,

Yes to see the data in the app both the bit in the advertise and the characteristics must be present.
For some algorithms ST requires to accept a license, in this case the bit advertise is added when you enable the algorithm adding the license.

This change is done firmware side when it detects a license and correctly initialize the library.

To enable the osxMotionFX algorithm (that enable you the quaternion and the compass) you can use the play store app or follow the guide that you find here:
http://www.st.com/content/st_com/en/products/embedded-software/mems-and-sensors-software/open-mems/osxmotionfx.html
and recompile your firmware.

Sorry but we are not open sourcing the app license manager.

Regards
Giovanni

GiovanniVisentini, 2017-02-17:
@adelorenz 
ok it seems that you are using it correctly.
Please, try to debug the node connection.
Set a breackpoint [here](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Android/blob/51339891c02357c506ce810d0d107ca874b2fb31/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L942) the function is called during the scanning phase, my featureMask is 0x4FF0400. Look if it builds the FeatureBattery class.

Then set a breakpoint [here](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Android/blob/51339891c02357c506ce810d0d107ca874b2fb31/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L313) The function is called during the node connection, look if it called with the BluetoothGattCharacteristics with uuid <code>00020000-0001-11e1-ac36-0002a5d5c51b</code> (the Battery characteristics)

Regards
Giovanni

dule-hwp, 2017-02-17:
Hi Giovanni, 

Thanks for the prompt response.

I deleted the previous comment after seeing that you actually published the play store app (without the license part though, but thats fine, i don't need that part anyway), so i wanted to browse the code first and see if i can find out how to enable that bit 6 in feature mask. However i still couldn't do it even with that code. But since you are saying its on the firmware side, i am not sure if that code would help me at all.

Anyway, i followed what you suggested and add osxMotionFX license, and this sets the bit number 8, which is expected (the one for Sensor Fusion Compact), and then i can see that feature in via node.getFeatures(). However, the Feature compass bit (number 6) is is still 0. Feature Mask: 0x4F40500 is printed in the android logcat (i would expect mask to be something like 0x4F40540). 

I am wondering can i somehow change the advertise bytes manually? Maybe using that config service with uuid: CONFIG_CONTROL_SERVICE_UUID. I see it has two characteristics FEATURE_COMMAND_UUID and REGISTERS_ACCESS_UUID. Is that the purpose of config service at all?

adelorenz, 2017-02-17:
Hi Giovanni,
at the first breakpoint I obtained the following featureClass values (in this order):

- `class com.st.BlueSTSDK.Features.FeatureAccelerationEvent`
- `class com.st.BlueSTSDK.Features.FeatureTemperature`
- `null`
- `class com.st.BlueSTSDK.Features.FeatureTemperature`
- `class com.st.BlueSTSDK.Features.FeatureHumidity`
- `class com.st.BlueSTSDK.Features.FeaturePressure`
- `class com.st.BlueSTSDK.Features.FeatureMagnetometer`
- `class com.st.BlueSTSDK.Features.FeatureGyroscope`
- `class com.st.BlueSTSDK.Features.FeatureAcceleration`
- `class com.st.BlueSTSDK.Features.FeatureMicLevel`

the `null` value caught my attention.

at the connection I obtained the following characteristics:
`001d0000-0001-11e1-ac36-0002a5d5c51b`
`00e00000-0001-11e1-ac36-0002a5d5c51b`
`00000400-0001-11e1-ac36-0002a5d5c51b`
`04000000-0001-11e1-ac36-0002a5d5c51b`
**`00020000-0001-11e1-ac36-0002a5d5c51b`**
`00000100-0001-11e1-ac36-0002a5d5c51b`
... omitted for the sake of brevity

GiovanniVisentini, 2017-02-17:
Correct, that null value is the problem..
could you see what value is returned by the method getDeviceId?
should be 0x02.

adelorenz, 2017-02-17:
Yes, it returns 2
`return this.mDeviceId;` _mDeviceId: 2_

GiovanniVisentini, 2017-02-20:
Are you using the last version of the sdk?
is this [line](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Android/blob/51339891c02357c506ce810d0d107ca874b2fb31/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Utils/BLENodeDefines.java#L308) present?

adelorenz, 2017-02-20:
Tnx Giovanni, I've updated to the last version and now it works!

closed 2017-02-20 by adelorenz