STMicroelectronics/BlueSTSDK_Android #5: Problem with real-time audio playback

cesarev91, 2017-06-10:
Ciao Giovanni,

I'm working on an app where one of the features allows sound reproduction of the sound received by SensorTile. I downloaded BlueSTSDK and I'm working directly on the BlueSTSDKExample module, adding activities and layouts.
I have a problem with extracting raw audio from the sample.
This is the code part of my activity where the problem arises:

`public final Feature.FeatureListener mAudioListener = new Feature.FeatureListener() {`
    `@Override`
    `public void onUpdate(final Feature f, final Feature.Sample sample) {`
         `audioSample = FeatureAudioADPCM.getAudio(sample);`
     `}`
`};`

Does not assign audio to audioSample.

Cordiali Saluti
Cesare

GiovanniVisentini, 2017-06-12:
Ciao Cesare,

your isuee is a bit generic..
What do you mean that the audio sample has not audio?
It is an empty array or as some noise on it?

- did you enable the feature notification?

- are you using also the FeatureAudioADPCMSync features?

- did you load the BlueVoice license form the ST BlueMS app?

Regards
Giovanni

cesarev91, 2017-06-12:
Sorry, you are right.
Empty array.
Yes for all three questions.

cesarev91, 2017-06-12:
This is the class for manage audio.

`package com.st.BlueSTSDK.Example;

import android.content.Context;
import android.content.Intent;
import android.media.AudioFormat;
import android.media.AudioManager;
import android.media.AudioTrack;
import android.os.Bundle;
import android.support.annotation.NonNull;
import android.support.v7.app.AppCompatActivity;
import android.view.View;
import android.widget.Button;
import android.widget.ImageButton;
import android.widget.SeekBar;

import com.st.BlueSTSDK.Feature;
import com.st.BlueSTSDK.Features.FeatureAudioADPCM;
import com.st.BlueSTSDK.Features.FeatureAudioADPCMSync;
import com.st.BlueSTSDK.Manager;
import com.st.BlueSTSDK.Node;
import com.st.BlueSTSDK.Utils.BVAudioSyncManager;

import java.util.List;

/**
 * Created by Cesare on 09/06/2017.
 */

public class FeatureAudioActivity extends AppCompatActivity {

    /**
     *   Node that will show the data
     */
    private Node mNode;

    /** fragment used for keep the connection open */
    private NodeContainerFragment mNodeContainer;

    //  Feature on which to apply the listener
    private FeatureAudioADPCM mAudio;

    // feature where we read the audio sync values
    private FeatureAudioADPCMSync mAudioSync;

    // The sampling rate
    private static final int SAMPLE_RATE = 8000;

    // raw audio
    private short audioSample[];

    // audio manager
    private static final int AUDIO_STREAM = AudioManager.STREAM_MUSIC;

    //  Audio track builder
    private AudioTrack mAudioTrack;

    //object containing the sync data needed in a ADPCM stream decoding
    private BVAudioSyncManager mBVAudioSyncManager = new BVAudioSyncManager();

    private final static String NODE_FRAGMENT =   FeatureAudioActivity.class.getCanonicalName() + "" +
            ".NODE_FRAGMENT";
    private final static String NODE_TAG = FeatureAudioActivity.class.getCanonicalName() + "" +
            ".NODE_TAG";

    /**
     * create an intent for start the activity that will log the information from the node
     *
     * @param c    context used for create the intent
[... 193 lines cut ...]
                public void onProgressChanged(SeekBar arg0, int progress, boolean arg2)
                {
                    mAudioManager.setStreamVolume(AudioManager.STREAM_MUSIC,
                            progress, 0);
                }
            });
        }
        catch (Exception e)
        {
            e.printStackTrace();
        }
    }

    /**
     * if we have to leave this activity, we force to keep the connection open, since we go back
     * in the {@link FeatureListActivity}
     */
    @Override
    public void onBackPressed() {
        mNodeContainer.keepConnectionOpen(true);
        super.onBackPressed();
    }//onBackPressed
}`

Sorry if I can not make it all like code but I'm new to both the programming and the forum

GiovanniVisentini, 2017-06-13:
Ciao Cesare,
Thanks for sharing the code, now the problem is more clear.

You are enabling the notification in the onCreate, and disabling it 6 line later.. so probably there are no time to send anything.. 
Put the enableNotification in the onResume and the disableNotification in the onPause.. 

You probably should call the mAudioTrack.write every time you have a new sample, otherwise you reproduce always the sample that was present when the user press the button. 

you can see an example [here](https://github.com/STMicroelectronics-CentralLabs/STBlueMS_Android/blob/c1b40f486202d96b3b191cb689a47e6ec95a5af0/BlueMS/src/main/java/com/st/BlueMS/demos/BlueVoiceFragment.java)

Regards
Giovanni

cesarev91, 2017-06-13:
Ciao Giovanni,
I tried to do as you said and now everything works!
Thanks so much

Regards
Cesare

closed 2017-06-15 by GiovanniVisentini