STMicroelectronics/BlueSTSDK_Android #7: Error during connection with node

cesarev91, 2017-07-05:
I have sold the SensorTile on the cradle because I can not receive data due to disconnection, both with my app and with the ST BlueMS app.
This problem also occurred when SensorTile was on the cradle expansion board , but it was not permanent, it happened 1 out of 6.

GiovanniVisentini, 2017-07-13:
Ok so there were some electric problems if the SensorTile was not solder on the cradle.
I reported your feedback to the hardware guys. 

Thanks
Giovanni

closed 2017-07-13 by GiovanniVisentini

cesarev91, 2017-07-13:
Hi Giovanni,

I've changed firmware and now it works.
First I used bluemicrosystem2 now use allmems1.
The only problem is that I have some disconnection as in a feature and then return to the function list. The SensorTile LED flashes four times (it means disconnected) and then reconnects (led off). The odd thing is that these disconnections do not always happen.