STMicroelectronics/BlueSTSDK_Android #8: Not working properly with Android 7

janakact, 2017-07-13:
I tried the demo application on Android 7 using Huawei GR5. It's not working properly. Sometimes device can't be detected. Or even if the device get detected it fails to connect. 
`E/com.st.BlueSTSDK.Node: Error connecting to the node:SensiBLE`

Sometimes it connects but app becomes really slow. After few seconds app stops receiving data. 

Tried the same app with Android it works there properly. Any idea how to fix this issue?

janakact, 2017-07-13:
Perhaps this will be helpful. 
https://blog.classycode.com/undocumented-android-7-ble-behavior-changes-d1a9bd87d983?gi=7aad2219d186

GiovanniVisentini, 2017-07-13:
Hi Janaka,

Thanks for the link but the scanning time out is set to 10 seconds so I don't think to be in that case.

Do you have other errors from the system before the error that you wrote?
Witch firmware are you using?

I tried with a SensiBLE and a Nexus9 with Android 7.1 and I didn't have problems. 
Tomorrow I will try with a Nexus 6 with Android 7.0

Best Regards
Giovanni

janakact, 2017-07-13:
I am not sure about the firmware version. How can I check?

GiovanniVisentini, 2017-07-14:
Hi Janank

I tried with the nexus6 and anroid 7 and I didn't have problems
About the fw version you can try to install the [BlueMS ](https://play.google.com/store/apps/details?id=com.st.bluems) app, connect (in this way we try a different apps based on the SDK and we see if it has the same problem) and select firmware update from the menu, if it is supported you should see the fw version.

By the way I think that is more something related to the device, could you paste 10-20 logcat line before the error?

Thanks

janakact, 2017-07-15:
Hi Giovanni,
It says `Firmware upgrade not available.`  Then goes back to demo view again so can't see the fw version. 

Thanks.

dinukasal, 2017-07-23:
Hi all,
I got the same error, `Firmware upgrade not available`

janakact, 2017-07-23:
Hi Giovanni, 
Here is the logcat. Sorry for the delay. 
```
07-23 14:00:27.654 28249-28249/? I/art: Late-enabling -Xcheck:jni
07-23 14:00:27.655 28249-28249/? I/art: Reinit property: dalvik.vm.checkjni= false
07-23 14:00:27.689 28249-28258/? E/art: Failed sending reply to debugger: Broken pipe
07-23 14:00:27.690 28249-28258/? I/art: Debugger is no longer active
07-23 14:00:27.690 28249-28258/? I/art: Starting a blocking GC Instrumentation
07-23 14:00:28.106 28249-28249/? W/System: ClassLoader referenced unknown path: /data/app/com.st.bluems.debug-2/lib/arm64
07-23 14:00:28.110 28249-28249/? W/art: JIT profile information will not be recorded: profile file does not exits.
07-23 14:00:28.143 28249-28249/? I/InstantRun: starting instant run server: is main process
07-23 14:00:28.241 28249-28249/? I/HwCust: Constructor found for class android.app.HwCustActivityImpl
07-23 14:00:28.292 28249-28249/? I/HwCust: Constructor found for class android.app.HwCustHwWallpaperManagerImpl
07-23 14:00:28.351 28249-28249/? W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
07-23 14:00:28.371 28249-28249/? I/BitmapFactory: set decoder allocator to gralloc
07-23 14:00:28.375 28249-28249/? I/gralloc: alloc_device_alloc:564: Alloc handle(0x7b847e3500): interfmt=1, stride=960, size=3686400, usage=0x2000003
07-23 14:00:28.402 28249-28249/? I/gralloc: alloc_device_alloc:564: Alloc handle(0x7b847e3700): interfmt=1, stride=832, size=2662400, usage=0x2000003
07-23 14:00:28.417 28249-28249/? I/gralloc: alloc_device_free:586: Free handle(0x7b847e3700)
07-23 14:00:28.417 28249-28249/? I/gralloc: alloc_device_free:586: Free handle(0x7b847e3500)
07-23 14:00:28.418 28249-28249/? E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:28.603 28249-28249/? E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:28.659 28249-28249/? I/HwSecImmHelper: mSecurityInputMethodService is null
07-23 14:00:28.665 28249-28249/? I/HwPointEventFilter: do not support AFT because of no config
07-23 14:00:28.759 28249-28307/com.st.bluems.debug I/OpenGLRenderer: Initialized EGL, version 1.4
07-23 14:00:28.775 28249-28249/com.st.bluems.debug W/art: Before Android 4.1, method int android.support.v7.widget.ListViewCompat.lookForSelectablePosition(int, boolean) would have incorrectly overridden the package-private method in android.widget.ListView
07-23 14:00:28.795 28249-28307/com.st.bluems.debug W/linker: /vendor/lib64/libhwuibp.so: unused DT entry: type 0xf arg 0xe3a
07-23 14:00:28.836 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 0
07-23 14:00:28.840 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 1
07-23 14:00:29.710 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 34359738371
07-23 14:00:29.721 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 240518168576
07-23 14:00:40.840 28249-28249/com.st.bluems.debug I/hwaps: JNI_OnLoad
07-23 14:00:41.000 28249-28249/com.st.bluems.debug E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:41.101 28249-28249/com.st.bluems.debug E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:41.167 28249-28249/com.st.bluems.debug I/HwPointEventFilter: do not support AFT because of no config
07-23 14:00:41.287 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 68719476744
07-23 14:00:41.312 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 68724719680
07-23 14:00:41.316 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 5242945
07-23 14:00:41.327 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 68719476736
07-23 14:00:43.438 28249-28249/com.st.bluems.debug I/BluetoothAdapter: getBluetoothLeScanner
07-23 14:00:45.448 28249-28249/com.st.bluems.debug I/com.st.BlueSTSDK.Node: Name: SensiBLE
                                                                                TxPower: 0
                                                                                Address: null
                                                                                Feature Mask: 0x5FE0398
                                                                                Protocol Version: 0x1
07-23 14:00:45.563 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 34359738369
07-23 14:00:49.906 28249-28249/com.st.bluems.debug E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:49.907 28249-28249/com.st.bluems.debug E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:49.909 28249-28249/com.st.bluems.debug E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:49.933 28249-28249/com.st.bluems.debug E/HW-JPEG-DEC: [HME_JPEG_DEC_Delete](3321): HME_JPEG_DEC_Delete: decoder_ctx=null
07-23 14:00:49.935 28249-28249/com.st.bluems.debug I/BitmapFactory: set decoder allocator to gralloc
[... cut ...]
07-23 14:01:41.495 28249-28255/com.st.bluems.debug I/art: After code cache collection, code=105KB, data=94KB
07-23 14:01:44.551 28249-28249/com.st.bluems.debug I/BluetoothAdapter: getBluetoothLeScanner
07-23 14:01:54.620 28249-28249/com.st.bluems.debug I/BluetoothAdapter: getBluetoothLeScanner
07-23 14:02:01.300 28249-28249/com.st.bluems.debug I/BluetoothAdapter: getBluetoothLeScanner
07-23 14:02:02.525 28249-28249/com.st.bluems.debug I/BluetoothAdapter: getBluetoothLeScanner
07-23 14:02:02.559 28249-28307/com.st.bluems.debug E/OpenGLRenderer: allen debug liyu Key: 103084458052
```

GiovanniVisentini, 2017-07-24:
Hi @dinukasal 

If you are using the SensBLE board and you got the `Firmware upgrade not available` message, is not a BlueMS problem. The app doesn't support the SensiBle board and we don't know witch firmware is running. 
Otherwise please open a new issue ([here](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android)) with more details (android version, mobile, board, Firmware version etc).

Hi @janakact 
I don't see error in the logs that you post.. the only strange things is the multiple call to `getBluetoothLeScanner` that is not directly used by the SDK. 
Maybe removing the deprecate api from the [Manager](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Android/blob/b490d4d382e6f8dddbf9ce6e02414a53d576cd58/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Manager.java) class will change something.

janakact, 2017-08-19:
It only happens with this Huawei GR5. I don't know why. seems like some kind of an optimization of their os.

MstrZeroOne, 2021-02-01:
I have the same problem on Huawei Y7 prime