STMicroelectronics/BlueSTSDK_Android #10: Passing nodes between activities

jcarellanov, 2017-08-24:
The app I'm working on will have to pass nodes between different activites. I was going to go with something like this, for Node n:

`Intent j = new Intent(ScanActivity.this, MainHub.class);
  j.putExtra("node",n);
  startActivity(j);`

But I realize now Node is not serializable since it has 
`private Handler mBleThread;`

I was wondering if you had any suggestions for retrieving the node object in MainHub?

GiovanniVisentini, 2017-08-25:
Hi,

you can pass the node tag (it is a string) and then get the node back using the Manager class.

if you need to keep the node connected, you can use the [NodeConnectionService](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android/blob/master/src/main/java/com/st/BlueSTSDK/gui/NodeConnectionService.java) and the [ActivityWithNode](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android/blob/master/src/main/java/com/st/BlueSTSDK/gui/ActivityWithNode.java) or the [NodeContainer](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android/blob/master/src/main/java/com/st/BlueSTSDK/gui/NodeContainer.java) interface.

you can see an example here: [DemoActivitiy](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android/blob/217cd7056e460004ab18f3eb48aedbd0295f5b51/src/main/java/com/st/BlueSTSDK/gui/DemosActivity.java#L610) passing the node to [DebugConsole](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android/blob/master/src/main/java/com/st/BlueSTSDK/gui/DebugConsoleActivity.java)

jcarellanov, 2017-08-27:
Thank you, that was it

closed 2017-08-27 by jcarellanov