STMicroelectronics/BlueSTSDK_Android #11: Pedometer unaccesible?

jcarellanov, 2017-08-27:
featurePedometer = mNode.getFeature(FeaturePedometer.class); always returns null.

I tried to get the steps through an AccelerationEvent listener, but getPedometerSteps always returns -1.
```
private Feature.FeatureListener AccelerometerSteps = new Feature.FeatureListener() {
        @Override
        public void onUpdate(Feature f, Feature.Sample sample) {
            if(featureAccelerationEvent!=null){
                steps = featureAccelerationEvent.getPedometerSteps(sample);
                if(steps != stepsOLD){
                    stepsOLD = steps;
                    MainActivity.this.runOnUiThread(new Runnable() {
                        @Override
                        public void run() {
                            number.setText(String.valueOf(steps));
                        }
                    });

                }
            }

        }
    };
```
All my other features and listeners are working properly. 

Thanks

GiovanniVisentini, 2017-08-28:
If the `getFeature` returns `NULL` it means that the data is not exported by the firmware..
Did you enable the license to use the pedometer? (see [1.13
Open.MEMS and Open.Audio license activation](http://www.st.com/resource/en/user_manual/dm00286083.pdf) )
To use the AccelerometerEvents did you call `detectEvent(PEDOMETER,true)` after enabling the notification?

Ps: if you upgrade to the [FP-SNS-ALLMEMS1](http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32-ode-function-pack-sw/fp-sns-allmems1.html) firmware the license is not needed anymore.

jcarellanov, 2017-08-28:
Thanks, I was missing the call to detectEvent(PEDOMETER,true). 

Incidentally, I see that enabling it means the other detectEvent I had becomes disabled. Is there any way to detect two events at once?

GiovanniVisentini, 2017-08-29:
Good,

with the current firmware/app is not possible. But we are thinking to change it since the MEMS driver supports multiple events. But I can't give you a precise schedule..

jcarellanov, 2017-08-29:
Hi,

Can I at least substitute the freefall event with the freefall feature? I
see that it's deprecated, what is the reason?

Thanks

GiovanniVisentini, 2017-08-29:
in the fw it was implemented in a different way, and with a different characteristics. 
I think that now the characteristics is not present anymore.
so basically you can't use it.

it is deprecated because the AccelerationEvent do more things and with a better implementation.

closed 2017-09-03 by jcarellanov