STMicroelectronics/BlueSTSDK_Android #12: Issue on Bluetooth paring with OnePlus and Huawei

fastlock, 2017-09-11:
Hello to everybody.
Every time i try to connect via Bluetooth with the board, i get an error or i cant even find it . 
The OCD console show me "CONNECTED ***MAC ADDRESS***" while the app tell me "LOST CONNECTION WITH THE NODE" or "ERROR DURING THE CONNECTION WITH THE NODE" (no stream of data between my board and my phone is detected after this point,even rebooting the phone).
The problem seems to affect only Oneplus and Huawei devices.  
Flushing Bluetooth data from the settings seems to work but the problem returns a connection after.
Ours smartphone work normally with other bluetooth devices.

GiovanniVisentini, 2017-09-19:
Hi Marco,

Thanks for the report just some other infos:
- Witch board/firmware are you using?
- could you report also the output of the log cat console?

fastlock, 2017-09-20:
I'm using the NucleoL476RG with the Example app and V2J28M18 firmware
this is my logcat: 

`09-20 15:26:12.349 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:26:16.427 31636-31636/com.st.BlueSTSDK.Example I/hwaps: JNI_OnLoad
09-20 15:26:19.141 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:26:21.418 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:26:28.332 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:26:33.593 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:26:56.770 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:26:58.221 31636-31636/com.st.BlueSTSDK.Example I/com.st.BlueSTSDK.Node: Name: ME1V310
                                                                                 	TxPower: 0
                                                                                 	Address: C0:83:53:35:3D:42
                                                                                 	Feature Mask: 0x20FD057B
                                                                                 	Protocol Version: 0x1
09-20 15:26:59.703 31636-31636/com.st.BlueSTSDK.Example E/FeatureList: loaded
09-20 15:26:59.885 31636-31636/com.st.BlueSTSDK.Example I/HwPointEventFilter: do not support AFT because of no config
09-20 15:26:59.895 31636-31636/com.st.BlueSTSDK.Example I/HwPointEventFilter: do not support AFT because of no config
09-20 15:26:59.903 31636-31636/com.st.BlueSTSDK.Example I/Choreographer: Skipped 34 frames!  The application may be doing too much work on its main thread.
09-20 15:26:59.978 31636-31636/com.st.BlueSTSDK.Example I/BluetoothDevice: connectGatt
09-20 15:27:06.886 31636-31636/com.st.BlueSTSDK.Example I/BluetoothAdapter: getBluetoothLeScanner
09-20 15:27:30.538 31636-31648/com.st.BlueSTSDK.Example E/com.st.BlueSTSDK.Node: Error connecting to the node:ME1V310
09-20 15:27:30.661 31636-31636/com.st.BlueSTSDK.Example I/HwPointEventFilter: do not support AFT because of no config
09-20 15:27:30.666 31636-31636/com.st.BlueSTSDK.Example I/BluetoothDevice: connectGatt
`