STMicroelectronics/BlueSTSDK_Android #13: Issue - State.Dead

IvanTerrito, 2018-03-29:
Thanks in advance.
My problem is that I am able to connect the Node, but after a few seconds in goes in a cycle of two states:

State --> Connecting, State --> Dead

And repeat... infinitely.
How could I to fix that?

GiovanniVisentini, 2018-03-29:
Hi Ivan,

Which board/Fw are you using?
What did you do after the node connection?
Do you have any message in the LogCat console?

Thanks
Giovanni

IvanTerrito, 2018-03-29:
I'm using a sample board: STEVAL-STLCX01V1

I'm going to copy here all my class. It is a Service:

`package com.it.baraka.services;

import android.Manifest;
import android.app.Service;
import android.bluetooth.BluetoothAdapter;
import android.content.Context;
import android.content.Intent;
import android.content.SharedPreferences;
import android.content.pm.PackageManager;
import android.location.LocationManager;
import android.os.Handler;
import android.os.IBinder;
import android.support.annotation.Nullable;
import android.support.v4.content.ContextCompat;
import android.util.Log;

import com.crashlytics.android.Crashlytics;
import com.it.baraka.NodeArrayAdapter;
import com.st.BlueSTSDK.Feature;
import com.st.BlueSTSDK.Features.FeatureAccelerationEvent;
import com.st.BlueSTSDK.Features.Field;
import com.st.BlueSTSDK.Manager;
import com.st.BlueSTSDK.Node;

import java.util.List;

/**
 * Created by ivanterrito on 15/02/17.
 */

public class BLUETOOTH_Service_Network extends Service {

    FeatureAccelerationEvent featureAccelerationEvent;

    //TODO COMMENT NEXT LINE
    int test = -2;

    public static final String TAG = "BLUET_Service_Network";

    Intent intent;
    public static final String BROADCAST_ACTION = "com.it.baraka.BLUETOOTH_Service_Network.getOrientation";

    private void sendOrientation() {
        Log.d(TAG, "entered sendOrientation");

        intent.putExtra("isConnected", isConnected);
        intent.putExtra("orientation", ORIENTATION);
        intent.putExtra("timestamp", O_TIMESTAMP);

        sendBroadcast(intent);
    }

    boolean gotFeatures = false;

    Handler handler;

    String ORIENTATION = "";
    String O_TIMESTAMP = "";

    String GESTURE = "empty";
    String ACCELEROMETER_EVENTS = "empty";
    String TEMPERATURE = "empty";
    String PRESSURE = "empty";
    String MAGNETOMETER = "empty";
    String GYROSCOPE = "empty";
    String ACCELEROMETER = "empty";
[... 350 lines cut ...]

        /*if(locationManagerNetwork != null){
            //noinspection MissingPermission
            locationManagerNetwork.removeUpdates(listenerNetwork);
        }*/

        //if (mGoogleApiClient.isConnected()) {
        //    LocationServices.FusedLocationApi.removeLocationUpdates(mGoogleApiClient, this);
        //    mGoogleApiClient.disconnect();
        //}
    }

    //SHARED PREFERENCES
    public static void putPref(String key, String value, Context context) {
        SharedPreferences prefs = context.getSharedPreferences("Settings", Context.MODE_PRIVATE);
        SharedPreferences.Editor editor = prefs.edit();
        editor.putString(key, value).apply();
    }

    public static String getPref(String key, Context context) {
        SharedPreferences preferences = context.getSharedPreferences("Settings", Context.MODE_PRIVATE);
        return preferences.getString(key, null);
    }

}`

GiovanniVisentini, 2018-03-29:
the BlueMicrosystem 2.2 is quite old firmware.. you should update it.

by the way could be that you call 2 times connect? (first time at: 03-29 14:51:13.379, second time at minute: 14:51:18.399)
the bluetooth stack is quite fragile, if you do 2 things at the time it breaks, and you have the generic status 133 ( 03-29 14:51:18.556 ).

The only way, that I know, to recover an status 133 is to avoid to go in that state :)

instead of using the handler with the self posting runnable you should use the listeners.

- create a manager listener that start the connection when find the correct node

- create a node listener that enable the notification when the node is connect and re connect if the node disconnect (like [here](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_GUI_Android/blob/e30bde79f9f412831ff97901831dc1848d35c4e4/src/main/java/com/st/BlueSTSDK/gui/NodeConnectionService.java#L165)

- create a feature listener that send the broadcast when the orientation change.

Said this, I don't know why your SensorTile disconnect and did not connect again in 5seconds.. 

Giovanni

IvanTerrito, 2018-03-29:
I do not try to connect in those lines. Those are SDK's Logs.
I check every 5 seconds for bluetooth, i need to reconnect every single time the bluetooth is ON, then i check if the mNode is != null and if it's not connected, ONLY in that case i look again for the device with mManager.getSharedInstance().getNodeWithTag("MY TAG DEVICE"); and call connect() function.

So I don't think the problem is on the flow... And yes the strangest thing is that it can't reconnect again, i must disconnect alimentation and reconnect again.

Do you have something more to suggest to me?

GiovanniVisentini, 2018-03-29:
look this line:

> 03-29 14:51:18.399 428-428/com.it.baraka D/NodeStateListener: BM2V220 Connecting-Connecting
03-29 14:51:18.399 428-428/com.it.baraka D/BLUET_Service_Network: onStateChange - Node.State: Connecting
03-29 14:51:18.399 428-428/com.it.baraka D/BLUET_Service_Network: onStateChange - Node.State: Connecting

do you know why are you printing 2 times the "onStateChange" line?

look to this line:
```JAVA
                 if(mNode != null) {
                            Log.d(TAG, "mNode != null");

                            if(!mNode.isConnected()) {
                                Log.d(TAG, "!mNode.isConnected()");

                                isConnected = true;

                                gotFeatures = false;

                                mNode.connect(getApplicationContext());

                                mNode.addNodeStateListener(new Node.NodeStateListener() {
                                    @Override
                                    public void onStateChange(Node node, Node.State newState, Node.State prevState) {
                                        Log.d(TAG,"onStateChange - Node.State: " + newState.name());

                                    }
                                });

                            }

```

your current state is "connecting" from this line:

> 03-29 14:51:13.379 428-428/com.it.baraka D/NodeStateListener: BM2V220 Dead->Connecting

so the isConnected return false, you call connect and you add a second listener that print the state change for the second time.. 

Can you use the BlueMS app or the example app and see if you have the same behaviours?

ps: I'm able to recognise the differences between logs and code lines ;)

IvanTerrito, 2018-03-29:
So.. should i declare the listener globally and add and remove later calling the variable, or just adding once?

The "problem" is that i need to keep on the time based handler.

I changed my code with the 1st case, calling remove and after add just for some tests, and it have again the same problems:

```
03-29 18:04:14.292 31601-31601/com.it.baraka D/BLUET_Service_Network: checkBluetooth()
03-29 18:04:14.293 31601-31601/com.it.baraka D/BLUET_Service_Network: Bluetooth is enabled
03-29 18:04:14.293 31601-31601/com.it.baraka D/BLUET_Service_Network: Paired tag: C0:7A:17:31:43:48
03-29 18:04:14.293 31601-31601/com.it.baraka D/BLUET_Service_Network: Paired name: BM2V220
03-29 18:04:14.293 31601-31601/com.it.baraka D/BLUET_Service_Network: mManager != null
03-29 18:04:14.295 31601-31601/com.it.baraka D/BLUET_Service_Network: mNode != null
03-29 18:04:14.296 31601-31601/com.it.baraka I/BLUET_Service_Network: Device already connected
03-29 18:04:14.296 31601-31601/com.it.baraka D/BLUET_Service_Network: entered sendOrientation
03-29 18:04:14.646 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.646 31601-31614/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.678 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.679 31601-302/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.694 31601-31613/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.694 31601-31613/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.771 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.782 31601-31613/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.782 31601-31613/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.784 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.784 31601-31614/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.787 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.787 31601-302/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.817 31601-31613/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.817 31601-31613/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:14.821 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:14.821 31601-31614/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:15.479 31601-31614/com.it.baraka D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=7 device=C0:7A:17:31:43:XX
03-29 18:04:15.494 31601-31614/com.it.baraka D/com.st.BlueSTSDK.Node: Node: BM2V220 Status: 8 newState: 0 boundState:10
03-29 18:04:15.494 31601-31614/com.it.baraka D/BluetoothGatt: close()
03-29 18:04:15.497 31601-31614/com.it.baraka D/BluetoothGatt: unregisterApp() - mClientIf=7
03-29 18:04:15.502 31601-31614/com.it.baraka E/com.st.BlueSTSDK.Node: Error connecting to the node:BM2V220
03-29 18:04:15.588 31601-31614/com.it.baraka D/NodeStateListener: BM2V220 Connected->Dead
03-29 18:04:15.588 31601-31614/com.it.baraka D/BLUET_Service_Network: onStateChange - Node.State: Dead
03-29 18:04:15.588 31601-31614/com.it.baraka D/NodeStateListener: BM2V220 Dead->Connecting
03-29 18:04:15.588 31601-31614/com.it.baraka D/BLUET_Service_Network: onStateChange - Node.State: Connecting
03-29 18:04:15.607 31601-31601/com.it.baraka D/BluetoothGatt: connect() - device: C0:7A:17:31:43:XX, auto: false
03-29 18:04:15.607 31601-31601/com.it.baraka D/BluetoothAdapter: isSecureModeEnabled
03-29 18:04:15.608 31601-31601/com.it.baraka D/BluetoothGatt: registerApp()
03-29 18:04:15.609 31601-31601/com.it.baraka D/BluetoothGatt: registerApp() - UUID=2ded7f1f-75b2-4d5c-9c80-66aa41dc19ad
03-29 18:04:15.715 31601-302/com.it.baraka D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
03-29 18:04:16.259 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.269 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.269 31601-31614/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:16.272 31601-31613/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.272 31601-31613/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:16.372 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.372 31601-302/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:16.380 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.384 31601-31613/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.384 31601-31613/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:16.389 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:16.389 31601-302/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:17.853 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:17.950 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:17.954 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:17.955 31601-31614/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:17.969 31601-31613/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:17.969 31601-31613/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:17.971 31601-302/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:17.982 31601-31614/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:17.982 31601-31614/com.it.baraka D/ScanRecord: first manudata for manu ID
[... cut ...]
03-29 18:04:25.901 31601-31613/com.it.baraka D/NodeStateListener: BM2V220 Connecting->Connecting
03-29 18:04:25.901 31601-31613/com.it.baraka D/BLUET_Service_Network: onStateChange - Node.State: Connecting
03-29 18:04:25.906 31601-4520/com.it.baraka D/ScanRecord: parseFromBytes
03-29 18:04:25.906 31601-4520/com.it.baraka D/ScanRecord: first manudata for manu ID
03-29 18:04:25.918 31601-31601/com.it.baraka D/BluetoothGatt: connect() - device: C0:7A:17:31:43:XX, auto: false
03-29 18:04:25.919 31601-31601/com.it.baraka D/BluetoothAdapter: isSecureModeEnabled
```

GiovanniVisentini, 2018-04-03:
I think you are still call connection when the node is connecting:
`03-29 18:04:19.306 31601-31601/com.it.baraka D/NodeStateListener: BM2V220 Connecting->Connecting`

can you use the BlueMS or the example app to see if the fw disconnect also with that applications?

Giovanni

GiovanniVisentini, 2018-04-03:
I see that you are calling the method `setEnable `the method should be protected, are you able to call it?

By the way it shouldn't have any effect. Look [here ](https://github.com/STMicroelectronics-CentralLabs/BlueSTSDK_Android/blob/4a983749b2ec19785cca3b4bbc0d898a1cc30542/BlueSTExample/src/main/java/com/st/BlueSTSDK/Example/FeatureListActivity.java#L275)to understand how to enable the notification

GiovanniVisentini, 2018-04-06:
try to enable swap the order of this lines: 
```JAVA
featureAccelerationEvent.detectEvent(FeatureAccelerationEvent.DetectableEvent.ORIENTATION, true);
mNode.enableNotification(featureAccelerationEvent);
```