STMicroelectronics/BlueSTSDK_Android #16: Feature List activity is not displaying the data from the SensorTile

PercyJacks, 2019-01-11:
Hi, I have downloaded your code and uploaded it to my device. However when the app connects to the SensorTile node, it only displays the timestamp and the "Status". I don't know if this is the problem but the status says "off". I am using the exact code from the repository with no changes so I don't know what the problem could be

GiovanniVisentini, 2019-01-15:
Hi Percy,

please give me more information to reproduce the problem:
1- witch firmware did you flash?
2- you connect to the sensor tile, an you correctly see the list of features
3- witch feature did you select?
4- you just see the timestamp (is it changing?) and the status that it is off, instead of the real data.

Thanks
Giovanni

PercyJacks, 2019-01-15:
Hi Giovanni, thank you for the quick response. It was actually a mistake in my programming of the SensorTile, once that was fixed, I was able to get values. 

Thank you
Percy

closed 2019-01-16 by GiovanniVisentini