STMicroelectronics/BlueSTSDK_Android #17: How can I make the SDK listen to a custom UUIDs on BLUEST board?

AviyaSmila, 2019-04-29:


GiovanniVisentini, 2019-04-29:
Hi try to see if the chapter 7 here:
https://www.st.com/content/ccc/resource/technical/document/user_manual/group1/ad/29/54/26/6a/ec/4c/8e/DM00550659/files/DM00550659.pdf/jcr:content/translations/en.DM00550659.pdf

answer to your question

AviyaSmila, 2019-05-01:
Does the SDK filter nodes that has unrelevant UUIDS?
What if I want to find all bluetooth devices?

GiovanniVisentini, 2019-05-02:
yes the SDK is filtering the devices using the advertise.
if you want to show other device you have to write your custom advertise filter, something like this one: [BlueNRGAdvertiseFilter.java](
https://github.com/STMicroelectronics/BlueSTSDK_GUI_Android/blob/62b5e246f5ba0d24b328d662d8b91f6b157544d8/src/main/java/com/st/BlueNRG/fwUpgrade/BlueNRGAdvertiseFilter.java)

and then pass it when you call the   startDiscovery as here: https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/0b772cbdb03a2fb346560708cb396e8a0297ee9d/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Utils/NodeScanActivity.java#L105

Best Regards

AviyaSmila, 2019-05-02:
How can I disable the completely the filtering?

GiovanniVisentini, 2019-05-02:
you can create an AdvertiseFilter class that always return an BleAdvertiseInfo != null, setting only the name and andres, and using fixed value for the other fields

AviyaSmila, 2019-05-02:
Thanks for the help Giovanni!

closed 2019-05-02 by GiovanniVisentini