STMicroelectronics/BlueSTSDK_Android #18: How can I get any incoming message(without using Feature class) and send it to external parser?

AviyaSmila, 2019-05-02:
How can I get any incoming message(without using Feature class) and send it to external parser?

GiovanniVisentini, 2019-05-02:
why you can't parse it inside the Feature?

by the way all the SDK is based on Feature, if you don't use it you can't enable notification/read characteristics ecc..

you can do a feature that copies all the data inside the Sample class end then do the parsing inside the onUpgrade method.. but the feature.extractData method exist to do the message parsing.. 

you can look to this class if you want to see something where not all parsing is done inside the extractData method: [FFTAmplitude](https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/0b772cbdb03a2fb346560708cb396e8a0297ee9d/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Features/FeatureFFTAmplitude.java)

AviyaSmila, 2019-05-02:
OK, how is the SDK map the incoming data to the relevant feature?

GiovanniVisentini, 2019-05-02:
yes exactly.
After that you register a the Feature class (mapping the UUID with the class that will manage it) you can forget the UUID the sdk will take care to see if it present and do the correct operation on it, and will present you the parsed data.

AviyaSmila, 2019-05-02:
Ok.
But how is this happening? 
Our BLE device isn`t ready yet and I want to get to know how the data processing is happening in the SDK

GiovanniVisentini, 2019-05-02:
✨  Magic  ✨  :stuck_out_tongue_winking_eye:
and something related to this code:
https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/0b772cbdb03a2fb346560708cb396e8a0297ee9d/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L520

AviyaSmila, 2019-05-02:
OK, I`m starting to get it...
another question - when I get the features of the node via the getfeatures() method the list dosent include the custom features which i`ve added..
how can I get those?

GiovanniVisentini, 2019-05-02:
you can request directly your feature with this:
[getFeature](https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/0b772cbdb03a2fb346560708cb396e8a0297ee9d/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L1440)

your custom char should be attached to the Node with this if:
https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/0b772cbdb03a2fb346560708cb396e8a0297ee9d/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L530

if your custom feature is not present:
1- you register it in a wrong way
2- the UUID is not present in the device
3- there is a bug in the sdk