STMicroelectronics/BlueSTSDK_Android #19: Random disconnects

AviyaSmila, 2019-07-03:
Hi, 
I`ve managed to connect to a custom device with the SDK but I have random disconnects that occur without any type pattern.
What can be the reason for this kind of problem?

GiovanniVisentini, 2019-07-03:
Hi,

what are you doing before the disconnection?
could be that you are sending/receiving too much data?
witch context do you use to connect, if the context die also your connection will die..
do you have any message from the log cat? (search for gatt and or bluetooth in the message tag, did not limit it to your application output) 

Best Regards
Giovanni

AviyaSmila, 2019-07-03:
D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=8 device=02:80:E1:00:00:AA
D/Bluetooth name:  is not null
D/com.st.BlueSTSDK.Node: Node: PG_1234567890 Status: 8 newState: 0 boundState:10
D/BluetoothGatt: close()
D/BluetoothGatt: unregisterApp() - mClientIf=8
D/Bluetooth name:  is not null
E/com.st.BlueSTSDK.Node: Error connecting to the node:PG_1234567890
D/Bluetooth name:  is not null
D/NodeStateListener: PG_1234567890 Connected->Dead
D/Connection Status: Dead
D/InputTransport: Input channel constructed: fd=69

This is the part where it disconnectss

GiovanniVisentini, 2019-07-03:
the status 8 should be a timeout error.. what did you do before the status change?

AviyaSmila, 2019-07-03:
sometimes it happens after sending data, sometimes after receiving data, sometimes it happens after doing nothing

AviyaSmila, 2019-07-07:
??

AviyaSmila, 2019-07-17:
@GiovanniVisentini ?

GiovanniVisentini, 2019-07-17:
try to add some breakpoint/ Log.d here:
https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/40fdcbf46955a025191589c6cff0e189bb17a2c9/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L1853

and here:
https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/40fdcbf46955a025191589c6cff0e189bb17a2c9/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L1598

to see if you have some write/enable notification pending that goes on timeout

another test could be to use another app that just connect to the node and see if the disconnection happen anyway, to see if it is a fw problem 

Best Regards
Giovanni

AviyaSmila, 2019-07-30:
I`ts not FW problem as other apps are keeping steady connection to the device