STMicroelectronics/BlueSTSDK_Android #20: Error on mNode.isConnected

RulerKun, 2019-07-19:
i seem to be getting error when i have reached the mNode.isConnected. The App just crash  
  
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean com.st.BlueSTSDK.Node.isConnected()' on a null object reference  
  
Please correct me if i am wrong but i need to use the ST BLE sensor in the playstore to upload firmware, or is there any other way?  
  
my board is STEVAL-MKSBOX1V1 (SensorTile.Box )   
  
sorry for the trouble

RulerKun, 2019-07-23:
@GiovanniVisentini i am actually trying to convert the listview to recyclerview  
  
but on the other question for the firmware, I actually have STLINK V3SET and STEVAL-MKSBOX1V1(SensorTile.Box), would like to ask if how can i flash the FP-Ai-sensing (https://my.st.com/content/my_st_com/en/products/embedded-software/mcu-mpu-embedded-software/stm32-embedded-software/stm32-ode-function-pack-sw/fp-ai-sensing1.html) or other firmware and change some details on what feature that can be shown in the android app. And as well if there is instructions/documentations on how to connect the SensorTile.Box and STLINK-V3SET

If there is any existing documentation, it would be great

GiovanniVisentini, 2019-07-24:
Hi,

The SensorTile.box and the STLink v3 are not compatible, they have different voltage. 
An adapter will be available in the next months.

The FP-AI-SENSING1 v3.0.0 has the bootloader that is not compatible with the default firmware.
so you have to use the ST Link v2 to flash the bootloader and the fw or  you can try to: 

1.  power off the board

2.  hold down the boot button while connect the USB cable

3.now the board is in DFU Mode, you can the STM32 Programmer  tool to connect to it

4. dump the flash to have a copy of the current fw 

5. flesh the new fw and bootloader

Best Regards

closed 2019-08-09 by RulerKun