STMicroelectronics/BlueSTSDK_Android #21: What is the Correct way of getting multiple information from Features

RulerKun, 2019-08-09:
```
    public Feature.FeatureListener mFeatureListener = new Feature.FeatureListener() {
        @Override
        public void onUpdate(@NonNull final Feature feature, @Nullable Feature.Sample sample) {
            if(mPressure){getPressure = String.valueOf(FeaturePressure.getPressure(sample));}
            if(mHumidity){getHumidity = String.valueOf(FeatureHumidity.getHumidity(sample));}
            if(mTemperature){getTemperature = String.valueOf(FeatureTemperature.getTemperature(sample));}
            if(mAccelerometer){ getAccX = String.valueOf(FeatureAcceleration.getAccX(sample));
                                getAccY = String.valueOf(FeatureAcceleration.getAccY(sample));
                                getAccZ = String.valueOf(FeatureAcceleration.getAccZ(sample));}
            if(mGyroscope){ getGyroX = String.valueOf(FeatureGyroscope.getGyroX(sample));
                            getGyroY = String.valueOf(FeatureGyroscope.getGyroY(sample));
                            getGyroZ = String.valueOf(FeatureGyroscope.getGyroZ(sample));}
            if(mMagnetometer){  getMagX = String.valueOf(FeatureMagnetometer.getMagX(sample));
                                getMagY = String.valueOf(FeatureMagnetometer.getMagY(sample));
                                getMagZ = String.valueOf(FeatureMagnetometer.getMagZ(sample));}
            if(mMic_Level){getMicLevel = String.valueOf(FeatureMicLevel.getMicLevel(sample,0));}
            if(mFFT_Amplitude){ getXComponent = String.valueOf(FeatureFFTAmplitude.getXComponent(sample));
                                getYComponent = String.valueOf(FeatureFFTAmplitude.getYComponent(sample));
                                getZComponent = String.valueOf(FeatureFFTAmplitude.getZComponent(sample));}

            mActivity.runOnUiThread(new Runnable() {
                @Override
                public void run() {
                    mFeatureInfoRecyclerView= (RecyclerView)mActivity.findViewById(R.id.st_page_featureinfo_recycler_view);
                    mActivity.runOnUiThread(new Runnable() {
                        @Override
                        public void run() {

                            featureinfolist.clear();
                            if(mPressure){featureinfolist.add("Pressure: "+getPressure);}
                            if(mHumidity){featureinfolist.add("Humidity: "+getHumidity);}
                            if(mTemperature){featureinfolist.add("Temperature: "+getTemperature);}
                            if(mAccelerometer){featureinfolist.add("Accelerometer X: "+getAccX +" Y: " + getAccY +" Z: " + getAccZ);}
                            if(mGyroscope){featureinfolist.add("Gyroscope X: "+getGyroX +" Y: " + getGyroY +" Z: " + getGyroZ);}
                            if(mMagnetometer){featureinfolist.add("Magnetometer X: "+getMagX +" Y: " + getMagY +" Z: " + getMagZ);}
                            if(mMic_Level){featureinfolist.add("Mic Level: "+getMicLevel);}
                            if(mFFT_Amplitude){featureinfolist.add("FFT Amplitude X: "+getXComponent +" Y: " + getYComponent +" Z: " + getZComponent);}
                            Log.d(TAG_PM,Arrays.toString(featureinfolist.toArray()));
                            if (firstloadinfo) {
                                mFeatureInfoRecyclerView.setLayoutManager(new LinearLayoutManager(mActivity) );
                                mBTFeatureInfoAdapter = new BTFeatureInfoAdapter(mActivity,featureinfolist);
                                mFeatureInfoRecyclerView.setAdapter(mBTFeatureInfoAdapter);
                                firstloadinfo=false;
                            }
                            else {
                                mBTFeatureInfoAdapter.notifyDataSetChanged();
                            }
                        }
                    });
                }
            });
        }
    };
```

[image]

I am using STM32F446RE nucleo for this, when i have tried this block of code to get the features needed and tried to putting them in a recyclerview, i am somehow getting errors because the values are jumping from one paramater to another.

What is the correct or better way to do this?

GiovanniVisentini, 2019-08-09:
Hi,

what are mPressure, mHumidity acc?

let's assume that you enable the notification for the pressure and the humidity, and use the same listener for both.

The onUpdate method will be called 2 times, one when a new pressure arrive and one when a new humidity arrive.
the getPressure/getHumidity just return the first value inside the Sample object. so if you call the getHumidity with a pressure sample you got a humidity of, for example, 1000 and no errors.. 
probably in your case both mPressure and mHumidity are true, and you update the getPressure and getHumidity string with the same values.

to fix it you can:
- define a listener for each feature [like here](https://github.com/STMicroelectronics/STBlueMS_Android/blob/5d3416185c0d9e5c44ada1401cdea1d8bb6d4be5/BlueMS/src/main/java/com/st/BlueMS/demos/EnvironmentalSensorsFragment.java#L359)

- use something like: `feature instanceof FeaturePressure` to understand witch data are inside the sample object [like here]( https://github.com/STMicroelectronics/STBlueMS_Android/blob/5d3416185c0d9e5c44ada1401cdea1d8bb6d4be5/BlueMSCloud/src/main/java/com/st/blesensor/cloud/AzureIoTCentral/AzureIoTCentralFeatureListener.java#L86 )

in this way you will update only the string associated with the feature that got updated.

Best Regards
Giovanni

RulerKun, 2019-08-09:
Hi, thanks for the reply.  
  

> what are mPressure, mHumidity acc?  

it is a boolean expression to see if the Feature is called  
  

> let's assume that you enable the notification for the pressure and the humidity, and use the same listener for both.

>The onUpdate method will be called 2 times, one when a new pressure arrive and one when a new humidity arrive.  

> probably in your case both mPressure and mHumidity are true, and you update the getPressure and getHumidity string with the same values.
  
yes, looks like calling the same listener for both function is wrong

Thank you very much for the support you have provided

RulerKun, 2019-08-14:
Hi @GiovanniVisentini 

would like to ask if there is a way to control the timing of Feature Listener?

GiovanniVisentini, 2019-08-14:
the notification are asynchronous, so the timing is controlled by the firmware.

what you can do is:
 - avoid the notification and read the feature value ( note that the characteristic needs to support the read operation)
- do a sub sampling implementing something like this: [SubSamplingFeatureListener](https://github.com/STMicroelectronics/STBlueMS_Android/blob/5d3416185c0d9e5c44ada1401cdea1d8bb6d4be5/BlueMSCloud/src/main/java/com/st/blesensor/cloud/util/SubSamplingFeatureListener.java#L8)

Best Regards
Giovanni 

ps: for different question please open different issue

RulerKun, 2019-08-14:
Thanks for the help!

sorry for not opening another issue, will try next time

closed 2019-08-14 by RulerKun

bloxofcodes, 2023-03-21:
@ArthurKun21 do you have sample repositrory of your app SAMPLE???

PezzoniL, 2023-03-21:
take a look here
https://github.com/STMicroelectronics/STBlueMS_Android/tree/FullAppSnapshot