STMicroelectronics/BlueSTSDK_Android #23: How to add a new feature?(UUID PROBLEM)

ANPuroOsso, 2020-01-22:
So I'm trying to add a new custom feature to the Android APP. I've already implemented the custom class that extends "Feature", have created the constructor having only the Node as a parameter, and have implemented the ExtractResult extractData method as well as a getter method for extracting data from the sample, as it's instructed in the documentation: https://github.com/STMicroelectronics/BlueSTSDK_Android

As far as step 2 goes, I'm not getting how do I get a "UUID" for this new feature and where exactly I have to add it in my code( the documentation says it has to be done before node connection, but I'm not sure where that would be)

ANPuroOsso, 2020-01-24:
@GiovanniVisentini  help!

ANPuroOsso, 2020-01-27:
up

GiovanniVisentini, 2020-01-28:
Hi,

in your code somewhere you will start the connection with your device calling the connect method.
before call that method create the ConnectionOption class as in the readme.

Best Regards
Giovanni

ANPuroOsso, 2020-01-28:
@GiovanniVisentini  Hello and thank you for your reply, I'm currently using the code for the ST BLE Sensor, but still I'm unsure as to where exactly it starts connecting, what I've done and it seems to have worked out is that I've written exactly what's here on page 23 https://www.st.com/content/ccc/resource/technical/document/user_manual/group1/ad/29/54/26/6a/ec/4c/8e/DM00550659/files/DM00550659.pdf/jcr:content/translations/en.DM00550659.pdf  on the onNodeDiscovered method, the problem is that by using "temp.append(0x00080000, MyFeature1.class);" the APP makes the new sensor take place of an older sensor and I wanted to avoid that. What I think would solve it was to discover exactly what's the UUID related to that new sensor I'm adding but I have no idea as to where that is

GiovanniVisentini, 2020-01-28:
yes you have to use the second example on that page.. 
add your code here:
https://github.com/STMicroelectronics/STBlueMS_Android/blob/3026be0ae48d49e8b8a81054dbb4b3a5ae4a625f/BlueMS/src/main/java/com/st/BlueMS/NodeListActivity.java#L82

ANPuroOsso, 2020-01-28:
Hmmm got it, the only thing I'm still not sure of is how do I generate a new UUID for that sensor, as it's not a sensor present in the  "section 6- Format of characteristics" https://www.st.com/content/ccc/resource/technical/document/user_manual/group1/ad/29/54/26/6a/ec/4c/8e/DM00550659/files/DM00550659.pdf/jcr:content/translations/en.DM00550659.pdf

GiovanniVisentini, 2020-01-28:
you can define a random UUID, there are site only for do it. 
than use the second example in the page 23

ANPuroOsso, 2020-01-28:
I did exactly what you said and generated a random UUID, but now my program is not recognizing my new feature in the getFeature method

GiovanniVisentini, 2020-01-28:
if you create a new UUID your firmware has to export the same uuid.. 
as I saw telling you here: https://github.com/STMicroelectronics/BlueSTSDK_Android/issues/24#issuecomment-579332345

ANPuroOsso, 2020-01-29:
I associated the custom feature with the UUID and my firmware is exporting the same UUID when sending that sensor's data but still the custom feature is not being recognized

GiovanniVisentini, 2020-01-29:
use the BLEToolbox to check if your firmware is exporting the uuid 
than set a breakpoint here:
https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/0b772cbdb03a2fb346560708cb396e8a0297ee9d/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Node.java#L521

and look why the feature is not build when your characteristics is selected.