STMicroelectronics/BlueSTSDK_Android #24: Sendig  different BLE characteristics

ANPuroOsso, 2020-01-28:
Hello I'm currently trying using the BLE Sensor APP by STM and I would like to work around the BLE 20 bytes cap when sending information.  Basically I want to know if the APP's structure given in this repository supports the receivement of two differente BLE packages, as in an example:  first my tile sends information of a given set of sensors and then, after the APP has received it the tile sends out information related to a different set of sensors

What's happening now is that I'm alternating bewtween sending package 1 with a set of sensors and package 2 with completely different sensors but what's happening is that package1's info is getting on where I showcase package 2 sensor's info and vice versa. Could the problem be in sending these 2 information packets with the same UUID or is it something that the frontend is doing wrong? And so, how can I fix it?

ANPuroOsso, 2020-01-28:
Would enabling notifications of certain sensors when I'm about to receive the package corresponding to them while desabling notifications of the  other package's sensors and vice versa work?

GiovanniVisentini, 2020-01-28:
the extractData method is called from a concurrent queue, so the is no guarantee about the order that it is executed, probably that is your problem.

you can use 2 different uuid and enable the notification on both.

ANPuroOsso, 2020-01-28:
@GiovanniVisentini Oh I see, I'm just a little confused about the "UUID", do you mean when the tile is sending characteristics there's a UUID representing the BLE package and I should make 2 differents UUID's for those mesages? If so, how exactly can I generate a new UUID to attach to this new BLE package?

GiovanniVisentini, 2020-01-28:
you can see the UUID as an address where read the data, you define 2 data location one for a set of sensors and the other for another set of sensors.
then you enable the notification on this 2 Feature and you will receive the notification on different classes.

ANPuroOsso, 2020-01-28:
@GiovanniVisentini  I see, so it's something primarily defined by the backend, the STM tile code responsible for sending the data?

GiovanniVisentini, 2020-01-28:
the STM Tile needs to export the UUID that you want to read, otherwise from the mobile you can read/receive notification..

ANPuroOsso, 2020-01-28:
@leonardoalencar71@yahoo.com.br

ANPuroOsso, 2020-01-28:
where in APP'S code does can I find the UUID exported by the tile?

GiovanniVisentini, 2020-01-28:
https://github.com/STMicroelectronics/BlueSTSDK_Android/blob/1f55866b370778a93afce98bba0dd011eb558747/BlueSTSDK/src/main/java/com/st/BlueSTSDK/Utils/BLENodeDefines.java#L239

but all the UUID listed there are already manage by the sdk..
and by the way the data exported depends to the firmware that you are running..

ANPuroOsso, 2020-01-28:
So what should I implement on the app's code to handle a different UUID?

GiovanniVisentini, 2020-01-28:
see here:
https://github.com/STMicroelectronics/BlueSTSDK_Android/issues/23

ANPuroOsso, 2020-01-28:
Sorry I still don't get it, the other thread I made was about adding external features not sending 2 different BLE packages

ANPuroOsso, 2020-01-28:
Does the tile have to send the UUID: "7e7419d5-cad5-4f01-91e2-2f3e5cd0a72c" when sending FeatureMyCustomClass data, is that what you're saying?
[image]

GiovanniVisentini, 2020-01-28:
the tile has to export the characteristics with that UUID.. 

you can use this sdk without knowing about uuid, and how the data are arriving to the mobile.. 
but if you want to do something custom it is better if understand how the BLE works.
otherwise we both lost time..