STMicroelectronics/BlueSTSDK_Android #29: Accelerometer event type not being recognized in android app (ST BLE Sensor)

Luktronics, 2022-03-04:
In the firmware I am sending the notification regarding the accelerometer event in this way:
         STORE_LE_16(NotifyAccEventCharData, 4000); //TimeStamp
         NotifyAccEventCharData[2]=0x08; // Tilt event
         STORE_LE_16(NotifyAccEventCharData+3 ,5000); //Steps
However, in the app (ST BLE Sensor) it shows the accelerometer event type being the value 264 (0x108) instead of 8 (0x08). The TimeStamp and Steps values are displayed correctly. What can I do to fix this problem?

saurabh-eegrab, 2023-02-01:
Hi @Luktronics ,
Hope you are doing good.
I just want to know whether you able to get Accelerometer data in your android application.
If yes , It will be great if you can share the solution.
Thanks

PezzoniL, 2023-03-01:
yes...
take a look on the plot demo where you can display the acc data that the
board is sending
if it's sending it.
and if you select the "start logging" pressing the the "3 dots" menu on the
top right of the screen....
it starts to saving what is receiving on .csv on your phone...
and when you press the "stop logging" the application will ask to you if
you want to share that .csv files using mail or something else

But remember
what you could see on android application is function of what the board is
sending using ble
Ciao
Luca