STMicroelectronics/BlueSTSDK_Android #31: Is there a way to use it in React-native App?

Dev-Aman7, 2023-07-22:


pulkit4603, 2025-05-05:
Did you ever find a way to use it in a react-native app?

PezzoniL, 2025-05-05:
Sorry but I had never used react-native
Ciao

Dev-Aman7, 2025-05-05:
> Did you ever find a way to use it in a react-native app?

There is an react ble library. It works with this.

Dev-Aman7, 2025-07-30:
I used react-native-ble-plx. But, I had to update multiple things in the node-modules to get it working which is not preferred. Expo has release it's own support for ble. We also migrated on that. Please use that.
https://expo.dev/blog/how-to-build-a-bluetooth-low-energy-powered-expo-app

closed 2025-07-30 by Dev-Aman7

PezzoniL, 2025-07-31:
about using the BlueST for audio streaming...
I had just released today a new version of the ST BLE Sensor application that uses the new version of the BlueST-SDK where I have updated (after some years) the opus library, compiling them for supporting the 16 KB page sizes.

If I will have time tomorrow, I will update this git with the latest modifications
the github repository for ST BLE sensor will be updated at the end of August
Ciao

Dev-Aman7, 2025-08-01:
It is not a good idea to stream audio on ble. It can send very less data. It is recommended to use classical bt for that.

I used to send some data from the sensor to app in the form of a string which is like in bytes.